Links2004/arduinoVNC #2: RANDOM_REG32' was not declared in this scope

alonewolfx2, 2016-03-10:
i am getting this issue on compiling. (arduino 1.6.7 and lastest arduino for esp8266)

```

C:\Users\alone\Documents\Arduino\libraries\arduinoVNC-master\src\VNC.cpp: In member function 'void arduinoVNC::vncRandomBytes(unsigned char*)':

C:\Users\alone\Documents\Arduino\libraries\arduinoVNC-master\src\VNC.cpp:1478:11: error: 'RANDOM_REG32' was not declared in this scope

     srand(RANDOM_REG32);
```

Links2004, 2016-03-10:
which ESP core version do you use?
2.1.0 will work fine.
not sure if 2.0.0 already have it.

alonewolfx2, 2016-03-12:
ok. i updated all libraries and core. its booting and semi-working. some colors wrong. propably its coming from my module. i will check on new module. thank you for your help.

closed 2016-03-12 by alonewolfx2

helpmeLINKS, 2020-09-24:
hey links i have this problem aswell.