Links2004/arduinoVNC #4: Question regarding possiblity of this library

shirish47, 2016-04-02:
Can we use same server and display on 4 different screens connected to 4 ESP8266? like cliping from 0,x/2 and 0,y/2  on one screen and x/2,x and y/2,y on other screen (similarly for lower part).

Also can we support other screens with SPI interface? you have used 2.2 inch display? any bigger display possible? like 2.8 or somthing or can we use some other processor to handle the SPI to other LCD interfaces?? just a thought. I know some kind of SPI to other interface would cause slow down.

Links2004, 2016-04-02:
the library has a Display interface class, if you implement the interface you can use any display you like.
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC.h#L141-L161
example implementation:
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC_ILI9341.cpp 

the protocol allow to ask the server only for a update of a specific area
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC.cpp#L809-L812
with some changes combining displays to view the data from one server is possible.

but you need to add a offset for x and y in the display class to write the right coordinates (but this is not a big problem thanks to the Interface class)
for this functions:
```cpp
display->draw_rect
display->draw_area
display->area_update_start
display->copy_rect
```

shirish47, 2016-04-03:
ok Great. That means sever would only send buffer of the display area we asked for. but if we have two displays asking for two different areas server would do that right?

Links2004, 2016-04-03:
yes no problem here.
each display have its one connection to the server and you can ask what you want in there.

shirish47, 2016-04-06:
thanks for the replay .. :)

closed 2017-10-11 by Links2004