Links2004/arduinoVNC #6: Is this library compatible with new ARM-based Arduino boards? (Zero, MKR1000, etc)

q2dg, 2017-06-05:
Thanks!!

Links2004, 2017-06-06:
not see any reason why not.
you may need to add a define for the correct network class name here (if there is a difference to the default ```EthernetClient```)
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC.h#L267
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC.h#L44

beside from this is normal CPP code, since the ESP8266 is already 32bit running on arm will work fine.

the display driver may be a bottle neck in therms of speed the adafruit once are not really fast,
for the ESP I have rewrite it 
https://github.com/Links2004/Adafruit_ILI9341
to get better FPS.
and added a new low level access to stream image data.

you can attache the VNC code to any display driver via the ```VNCdisplay``` class.
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC.h#L141
example:
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC_ILI9341.cpp

q2dg, 2017-06-06:
Thanks a lot.
I'll try it and comment here.
Thanks!

closed 2017-06-06 by q2dg