Links2004/arduinoVNC #9: Scramble Display on TightVNC

mltongca, 2017-10-09:
The display is sort of working with a bit scramble when TightVNC server is connected.

Links2004, 2017-10-09:
most likely a HW problem at 80Mhz the connections an cables starts to matter.
you can try to change it to 40Mhz and see how it goes, if its fine then, you need to improve you wirering or you display can not handle 80Mhz well.

https://github.com/Links2004/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.cpp#L267

mltongca, 2017-10-11:
SPI.setFrequency(60000000) function is used to change the SPI clock to 60Mhz and it is working fine now. Thank you.

closed 2017-10-11 by Links2004

Links2004, 2017-10-11:
Note:
the ESP can not run the SPI at 60Mhz it will run at 40Mhz when you call the setFrequency with 60Mhz.
this is a Hardware limitation, based on how the frequency divider works internally.