Links2004/arduinoVNC #10: Skipping Video Frame

mltongca, 2017-10-11:
When running TightVNC "single window sharing" 720p video on Windows Media Player, some frames are skipped. Windows TightVNC client is working fine.

Links2004, 2017-10-11:
there is a limit on SPI and Network speed, I where able to reach 20-25 fps @ 320x240 
with HEXTILE encoding and SPI @ 80Mhz and the ESP core at 160Mhz.

fps has some variance based on how compressed the data is.
the ESP requests new frames as fast as it can process them.
If you speed up the core the FPS goes up a bit, but SPI speed makes the most difference.
since for every full frame 153600 Bytes need to be send to the display, at 80Mhz this takes around 15ms.

if the ESP only need to send out data the max frame rate is about 65fps,
but since we need to handle WiFi encryption, Network decoding, VNC decoding, and the SPI Protokoll for the display on one single core. 
the results are surprisingly good.

you windows client has much more power the ESP runs default at 80Mhz but can be set to 160Mhz.
but its still a big difference ;)

mltongca, 2017-10-11:
Thank you for your explanation. Would FPS performance improve using ESP32?

Links2004, 2017-10-12:
sure, its dual core and HW AES for the WiFi.
but I never tried to run the lib on a ESP32.
since the lib only uses the Network class,
and the display only needs SPI i dont see a reason why it will not run there.