Links2004/arduinoVNC #11: Scaling

njordan77, 2017-12-30:
Sorry to write this as an issue, but did not see any different way to get in touch.
Is there any way to scale a 640x480 remote session on a 2,8inch 320x240 display....I can connect to remote server (NodeMCU+ILI9341)...to my heating system. BUT the image is completely messed up.
Guess this is because of the pixel mismatch. Could not find any way to change/scale the incoming resolution.....is there anything possible? as i do not know of any ILI9341 displays that would be handling 640x480.

Cheers, Norbert

Links2004, 2017-12-31:
Scaling needs to be done on the server side, the ESP has to less power for this.
most server support ```geometry``` as option.

the lib only requests the 320x240 but some servers have a bad implementation
and always send the full image regardless of what is requested via the protocol.
this is most likely why you image is messed up.
if the server is correct implemented you will get the 320x240 pixel from the top left 
if the VNC server geometry is bigger then the 320x240.

https://github.com/Links2004/arduinoVNC/blob/master/src/VNC.cpp#L809
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC.cpp#L667

to get the full image the best you can do is set the geometry on the server correct.

for linux this may help:
https://github.com/Links2004/arduinoVNC/issues/3

njordan77, 2018-01-06:
Hi, thanks for your feedback and ideas.

What makes me sceptical about your scaling speculation is that the server (a heating with linux-based OS) works OK with a Raspi Zero and a 480x320 display and the server display is a 640x480 one...so for RealVNC viewer @Stretch Linux it works to scale down....as you said that scaling is done at server side.

Is there anything else I could change in the source code to force something (?)