Links2004/arduinoVNC #17: HEY LINKS HELP MEPLEASE

helpmeLINKS, 2020-09-23:
hello, can anyone help me? im new to coding and I can't figure out how to compile. Im sure that i have the board in the right settings and the right port. I have all the right libraries downloaded. i have a good wire and 3.0 usb plug in. i dont know man HELP ME!!!!!!!

/*
   VNC_ILI9341.ino

    Created on: 07.01.2015

   required librarys:
    - SPI (arduino core)
    - WiFi (arduino core)
    - Adafruit_GFX (https://github.com/adafruit/Adafruit-GFX-Library)
    - Adafruit_ILI9341 (https://github.com/Links2004/Adafruit_ILI9341)
    - arduinoVNC (https://github.com/Links2004/arduinoVNC)
*/

#include <Arduino.h>
#ifdef ESP8266
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#endif
#include <SPI.h>

#include <Adafruit_I2CDevice.h>
#include <Adafruit_GFX.h>

#include <VNC_ILI9341.h>
#include <VNC.h>

// ILI9341
#define TFT_DC      (5)
#define TFT_CS      (15)
#define TFT_RESET   (4)
// SPI:
// SCK to 14 (18 on esp32)
// MISO to 12 (19 on esp32)
// MOSI to 13 (23 on esp32)

const char * vnc_ip = "192.168.1.12";
const uint16_t vnc_port = 5900;
const char * vnc_pass = "12345678";

const char* ssid = "xxxx";
const char* password = "xxx";

ILI9341VNC tft = ILI9341VNC(TFT_CS, TFT_DC, TFT_RESET);
arduinoVNC vnc = arduinoVNC(&tft);

void TFTnoWifi(void) {
  tft.fillScreen(ILI9341_BLACK);
  tft.setCursor(0, ((tft.getHeight() / 2) - (5 * 8)));
  tft.setTextColor(ILI9341_RED);
  tft.setTextSize(5);
  tft.println("NO WIFI!");
  tft.setTextSize(2);
  tft.println();
}

void TFTnoVNC(void) {
  tft.fillScreen(ILI9341_BLACK);
  tft.setCursor(0, ((tft.getHeight() / 2) - (4 * 8)));
  tft.setTextColor(ILI9341_GREEN);
  tft.setTextSize(4);
  tft.println("connect VNC");
  tft.setTextSize(2);
  tft.println();
  tft.print(vnc_ip);
  tft.print(":");
  tft.println(vnc_port);
}

void setup(void) {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  // Init ILI9341
  tft.begin();
  delay(10);
  tft.setRotation(1);

  tft.fillScreen(ILI9341_BLUE);
#ifdef ESP8266
  // disable sleep mode for better data rate
  WiFi.setSleepMode(WIFI_NONE_SLEEP);
#endif
  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);
  TFTnoWifi();
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  TFTnoVNC();

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.println(F("[SETUP] VNC..."));

  vnc.begin(vnc_ip, vnc_port);
  vnc.setPassword(vnc_pass); // optional
}

void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    vnc.reconnect();
    TFTnoWifi();
    delay(100);
  } else {
    vnc.loop();
    if (!vnc.connected()) {
      TFTnoVNC();
      // some delay to not flood the server
      delay(5000);
    }
  }
}

ERROR MESSAGE

Arduino: 1.8.14 Hourly Build 2020/09/14 03:33 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Multiple libraries were found for "Adafruit_ILI9341.h"

 Used: C:\Users\Herre\OneDrive\Documents\Arduino\libraries\Adafruit_ILI9341

 Not used: C:\Users\Herre\OneDrive\Desktop\arduino-nightly\libraries\Adafruit_ILI9341-master

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Herre\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi 

Using library SPI at version 1.0 in folder: C:\Users\Herre\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SPI 

Using library Adafruit_BusIO at version 1.5.0 in folder: C:\Users\Herre\OneDrive\Documents\Arduino\libraries\Adafruit_BusIO 

Using library Wire at version 1.0 in folder: C:\Users\Herre\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\Wire 

Using library Adafruit_GFX at version 1.10.1 in folder: C:\Users\Herre\OneDrive\Documents\Arduino\libraries\Adafruit_GFX 

Using library arduinoVNC-1.1.0 at version 1.1 in folder: C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC-1.1.0 

Using library Adafruit_ILI9341 at version 1.5.6 in folder: C:\Users\Herre\OneDrive\Documents\Arduino\libraries\Adafruit_ILI9341 

exit status 1

Error compiling for board Generic ESP8266 Module.

Links2004, 2020-09-23:
Hi,

currently you have Multiple `Adafruit_ILI9341` libraries.

paths:
- C:\Users\Herre\OneDrive\Documents\Arduino\libraries\Adafruit_ILI9341
- C:\Users\Herre\OneDrive\Desktop\arduino-nightly\libraries\Adafruit_ILI9341-master

removing one is the first step.

if you still have problems may go back to a stable Arduino build and try again.
may enable verbose output in the Preferences to see more.

helpmeLINKS, 2020-09-24:
hello,
I reinstalled Arduino and the proper libraries. still contines to fail. thank you.

C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp: In member function 'virtual void ILI9341VNC::draw_area(uint32_t, uint32_t, uint32_t, uint32_t, uint8_t*)':
C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp:53:5: error: 'area_update_start' is not a member of 'Adafruit_ILI9341'
     Adafruit_ILI9341::area_update_start(x, y, w, h);
     ^
C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp:54:5: error: 'area_update_data' is not a member of 'Adafruit_ILI9341'
     Adafruit_ILI9341::area_update_data(data, (w*h));
     ^
C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp:55:5: error: 'area_update_end' is not a member of 'Adafruit_ILI9341'
     Adafruit_ILI9341::area_update_end();
     ^
C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp: In member function 'virtual void ILI9341VNC::area_update_start(uint32_t, uint32_t, uint32_t, uint32_t)':
C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp:68:5: error: 'area_update_start' is not a member of 'Adafruit_ILI9341'
     Adafruit_ILI9341::area_update_start(x, y, w, h);
     ^
C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp: In member function 'virtual void ILI9341VNC::area_update_data(char*, uint32_t)':
C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp:72:5: error: 'area_update_data' is not a member of 'Adafruit_ILI9341'
     Adafruit_ILI9341::area_update_data((uint8_t *)data, pixel);
     ^
C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp: In member function 'virtual void ILI9341VNC::area_update_end()':
C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC\src\VNC_ILI9341.cpp:76:5: error: 'area_update_end' is not a member of 'Adafruit_ILI9341'
     Adafruit_ILI9341::area_update_end();
     ^
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Herre\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi 
Using library SPI at version 1.0 in folder: C:\Users\Herre\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SPI 
Using library Adafruit_GFX_Library at version 1.10.1 in folder: C:\Users\Herre\OneDrive\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Adafruit_ILI9341-master at version 1.5.6 in folder: C:\Users\Herre\OneDrive\Documents\Arduino\libraries\Adafruit_ILI9341-master 
Using library arduinoVNC at version 1.1 in folder: C:\Users\Herre\OneDrive\Documents\Arduino\libraries\arduinoVNC 
Using library Adafruit_BusIO at version 1.5.0 in folder: C:\Users\Herre\OneDrive\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 1.0 in folder: C:\Users\Herre\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\Wire 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Links2004, 2020-09-24:
the current error comes from the fact that you are not using my version of the `Adafruit_ILI9341` 
https://github.com/Links2004/Adafruit_ILI9341
I did many speed optimizations and changes to for the ESP8266 in `Adafruit_ILI9341`.