Links2004/arduinoVNC #18: ST7789 support?

misaalanshori, 2020-11-05:
Hello, I want to use this on my ESP8266, but I don't know how or if it's possible to use an ST7789 240x240 SPI display. It says on the readme "more possible using VNCdisplay Interface" but I don't really understand what it means. I don't really have much knowledge on arduino and c++

Links2004, 2020-11-09:
the VNCdisplay interfaces allows you to plug in any display / output you want.

what is needed is the implementation of the `VNCdisplay` class.
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC.h#L143-L163

example:
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC_ILI9341.cpp

modi12jin, 2021-01-11:
Does this library support TFT_eSPI?

Links2004, 2021-01-11:
currently not, bit after a short look at  the lib it looks like the needed functions are there.
mostly:
https://github.com/Bodmer/TFT_eSPI/blob/5f171eeefdb8674ee85e5dde26dcee955524510a/TFT_eSPI.h#L394
https://github.com/Bodmer/TFT_eSPI/blob/5f171eeefdb8674ee85e5dde26dcee955524510a/TFT_eSPI.h#L418
https://github.com/Bodmer/TFT_eSPI/blob/5f171eeefdb8674ee85e5dde26dcee955524510a/TFT_eSPI.h#L384-L385

what is needed is the implementation of the VNCdisplay class.
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC.h#L143-L163
which comes done to calling the right functions of the TFT_eSPI.

example for ILI9341
https://github.com/Links2004/arduinoVNC/blob/master/src/VNC_ILI9341.cpp

modi12jin, 2021-01-12:
Does Adafruit-ST7735-Library support?

modi12jin, 2021-01-17:
@Links2004 

> I use ST7789 on ESP32 and there is a problem of color mismatch

[image]

> VNC_STTT89.cpp

```
#include "VNC_config.h"

#include "VNC.h"

#include "TFT_eSPI.h"
#include "VNC_ST7789.h"

ST7789VNC::ST7789VNC(){
    TFT_eSPI();
}

bool ST7789VNC::hasCopyRect(void) {
    return false;
}

uint32_t ST7789VNC::getHeight(void) {
    return 240;
}

uint32_t ST7789VNC::getWidth(void) {
    return 240;
}

void ST7789VNC::draw_area(uint32_t x, uint32_t y, uint32_t w, uint32_t h, uint8_t *data) {
    TFT_eSPI::pushImage(x, y, w, h, (uint16_t*)data);
}

void ST7789VNC::draw_rect(uint32_t x, uint32_t y, uint32_t w, uint32_t h, uint16_t color) {
    TFT_eSPI::fillRect(x, y, w, h, ((((color) & 0xff) << 8) | (((color) >> 8))));
}

void ST7789VNC::copy_rect(uint32_t src_x, uint32_t src_y, uint32_t dest_x, uint32_t dest_y, uint32_t w, uint32_t h) {

}

void ST7789VNC::area_update_start(uint32_t x, uint32_t y, uint32_t w, uint32_t h) {
    TFT_eSPI::setAddrWindow(x, y, w, h);
}

void ST7789VNC::area_update_data(char *data, uint32_t pixel){
    TFT_eSPI::pushPixels((uint8_t *)data, pixel);
}

void ST7789VNC::area_update_end(void){
//#ifndef ESP32
//    Adafruit_ILI9341::area_update_end();
//#endif
}
```

> VNC_ST7789.h

```
#ifndef VNC_ST7789_H_
#define VNC_ST7789_H_

#include "VNC_config.h"
#include <TFT_eSPI.h>
#include "VNC_ST7789.h"
#include "VNC.h"

class ST7789VNC: public VNCdisplay, public TFT_eSPI {
    public:
        ST7789VNC();

        bool hasCopyRect(void);

        uint32_t getHeight(void);
        uint32_t getWidth(void);

        void draw_area(uint32_t x, uint32_t y, uint32_t w, uint32_t h, uint8_t *data);

        void draw_rect(uint32_t x, uint32_t y, uint32_t w, uint32_t h, uint16_t color);

        void copy_rect(uint32_t src_x, uint32_t src_y, uint32_t dest_x, uint32_t dest_y, uint32_t w, uint32_t h);

        void area_update_start(uint32_t x, uint32_t y, uint32_t w, uint32_t h);
        void area_update_data(char *data, uint32_t pixel);
        void area_update_end(void);

    private:
        uint32_t area_x, area_y, area_w, area_h;

};

#endif
```

> main.cpp

```
#include <TFT_eSPI.h>

#include "VNC_ST7789.h"
#include <VNC.h>

const char * vnc_ip = "192.168.0.105"; //VNC Server IP
const uint16_t vnc_port = 5900;
const char * vnc_pass = ""; //No password

const char* ssid = "FAST_BA74";
const char* password = "xxxxxxxxx";

ST7789VNC tft = ST7789VNC();
arduinoVNC vnc = arduinoVNC(&tft);

void setup(void) {
    tft.begin();
    Serial.begin(115200);

    tft.setRotation(0);
    tft.setTextSize(2);
    tft.setCursor(0, 0, 1);
    tft.println("Ready");

    Serial.setDebugOutput(true);
    Serial.println();
    Serial.println();
    Serial.println();

    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);
    while(WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    Serial.println(F("[SETUP] VNC..."));

    vnc.begin(vnc_ip, vnc_port);
    //vnc.setPassword(vnc_pass); // check for vnc server settings
}

void loop() {
    if(WiFi.status() != WL_CONNECTED) {
        vnc.reconnect();
        delay(100);
    } else {
        vnc.loop();
        if(!vnc.connected()) {
            delay(5000);
        }
    }
}
```

Links2004, 2021-01-17:
this looks like the `ST7789` does not use the same RGB encoding the any display I had in hand.
you most likely can fix this by changing the color bit order here:

https://github.com/Links2004/arduinoVNC/blob/92731449d7f2333bc4f2a56a49f3081b5679d222/src/VNC.cpp#L113-L119

current encoding:

Rot: 5 Bit MSB
Grün: 6 Bit
Blau: 5 Bit LSB

may changing the endianes can help to:

https://github.com/Links2004/arduinoVNC/blob/92731449d7f2333bc4f2a56a49f3081b5679d222/src/VNC.cpp#L106-L110

if this fixes it we can adapt the lib to allow configuring this on a by display bases.

modi12jin, 2021-01-17:
@Links2004 

Color is normal

[image]

I only modify this code

> VNC.cpp

```
#ifdef ESP32
    opt.client.bigendian = 1;
#else
    opt.client.bigendian = 1;
#endif
```

Links2004, 2021-01-17:
please check if the `ST7789` branch does work, if yes I will release a new version of the lib.

https://github.com/Links2004/arduinoVNC/tree/ST7789
https://github.com/Links2004/arduinoVNC/pull/19

modi12jin, 2021-01-17:
@Links2004 

normal operation

[image]

Links2004, 2021-01-17:
thanks for testing, version 1.4 release with ST7789 support

modi12jin, 2021-01-19:
@Links2004 

How to start DMA to increase the number of screen frames?

Links2004, 2021-01-20:
that depends on the ST7789 library supporting DMA for the SPI.
in the VNC protocol side there is no use for DMA since the Image data is encoded and we need CPU logic to decode the protocoll.
the data transfer from the Image buffer to the ST7789 may can use DMA for SPI but that depands on the ST7789 TFT_eSPI lib.
what comes down to how the TFT_eSPI functions are implemented in https://github.com/Links2004/arduinoVNC/blob/master/src/VNC_ST7789.cpp

Aleyhanz, 2023-06-18:
Hallo .i have a lcd tft ili9341 parallel 8bit.can you help me make a library for this lcd..thank you before