Links2004/arduinoVNC #20: VGA display support using FabGL

niutech, 2021-02-23:
Could arduinoVNC support VGA display and input events using [FabGL](https://github.com/fdivitto/FabGL) library on one of its [compatible boards](https://github.com/fdivitto/FabGL/wiki/Boards)?

Links2004, 2021-02-23:
looks possible with `writeScreen` http://www.fabglib.org/classfabgl_1_1_v_g_a_controller_aab70542bc1007b141bb2d6db3acbce02.html#aab70542bc1007b141bb2d6db3acbce02

the for the output the implementation of `VNCdisplay` is needed, see
- VNC_ILI9341.cpp
- VNC_ILI9341.h
or
- VNC_ST7789.cpp
- VNC_ST7789.h
for examples

mouse event example:
https://github.com/Links2004/arduinoVNC/blob/8a5cd70dd8d9661906dc9040d4e2ea57b20a682d/src/VNC.h#L187
https://github.com/Links2004/arduinoVNC/blob/8a5cd70dd8d9661906dc9040d4e2ea57b20a682d/examples/VNC_ILI9341_touch/VNC_ILI9341_touch.ino#L100-L121

keyboard is possible but there is no public function Implement yet, but easy possible with using:
https://github.com/Links2004/arduinoVNC/blob/8a5cd70dd8d9661906dc9040d4e2ea57b20a682d/src/VNC.cpp#L1009-L1017

niutech, 2021-02-23:
Thank you very much!

closed 2024-05-31 by TheNitek