Links2004/arduinoVNC #21: Is sound supported?

NonaSuomy, 2021-03-06:
Is audio streaming supported by this?

Thank you.

Links2004, 2021-03-06:
no, the VNC protocol does not support sound.

NonaSuomy, 2021-03-06:
https://www.realvnc.com/en/connect/audio/

https://help.realvnc.com/hc/en-us/articles/360002504358-Audio-FAQ#how-does-the-audio-feature-work--0-9

https://guacamole.apache.org/doc/gug/configuring-guacamole.html#vnc

enable-audio | If set to "true", audio support will be enabled, and a second connection for PulseAudio will be made in addition to the VNC connection. By default, audio support within VNC is disabled.

audio-servername | The name of the PulseAudio server to connect to. This will be the hostname of the computer providing audio for your connection via PulseAudio, most likely the same as the value given for the hostnameparameter.If this parameter is omitted, the default PulseAudio device will be used, which will be the PulseAudio server running on the same machine as guacd.

Links2004, 2021-03-06:
PulseAudio is independent of the VNC protocol.
its basically a seconde connection with its own protocol.

in the realvnc cases its a prosperity change they made behind a pay wall, so hard to say how they do it but its not part of the RFC for VNC for sure.

more details about VNC and the TFB protocol can be found here:
https://github.com/Links2004/arduinoVNC/tree/master/docu
https://tools.ietf.org/html/rfc6143

if you want to stream via PulseAudio you need a second lib that handles this.
but the ESP8266 more then likely has to less CPU power to handle VNC and audio at the same time.
with two ESP8266 one for VNC and one for PulseAudio this may is possible.
Alternative experiment with the ESP32 and its two CPU cores.

closed 2021-03-29 by Links2004