Links2004/arduinoVNC #22: rfbProcessClientNormalMessage: ignoring unknown encoding type -313

llandsmeer, 2021-03-28:
Hi Links2004,

Thanks for making this incredible useful project available. I tried to get a sketch working using arduinoVNC on my M5Paper (esp32), but it seems like it has trouble receiving screen updates.

The [sketch](https://github.com/Links2004/arduinoVNC/files/6218144/sketch.txt) I created is attached. For the moment, I'm just listening in on calls to my `VNCdisplay`, sharing my Ubuntu gnome display through the builtin `vino-server` with authentication disabled. Getting a connection works, and I receive a notification every time I copy a piece a text. Sadly, I don't receive screen updates, and I think its related to this error: **rfbProcessClientNormalMessage: ignoring unknown encoding type -313**

I'm not familiar with VNC internals, but a quick scan through the RFC didn't show -313 as a normal encoding indeed. How do I fix this error and/or get screen updates to my eInk screen? :smile:

When I get it working, I'll happily create a pull request for my VNCdisplay implementation if you think others might benefit from it

Kind regards,
Lennart Landsmeer

Full vino-server logs below:

```
28/03/2021 23:53:51 Autoprobing TCP port in (all) network interface
28/03/2021 23:53:51 Listening IPv6://[::]:5900
28/03/2021 23:53:51 Listening IPv4://0.0.0.0:5900
28/03/2021 23:53:51 Autoprobing selected port 5900
28/03/2021 23:53:51 Advertising security type: 'TLS' (18)
28/03/2021 23:53:51 Re-binding socket to listen for VNC connections on TCP port 5900 in (all) interface
28/03/2021 23:53:51 Listening IPv6://[::]:5900
28/03/2021 23:53:51 Listening IPv4://0.0.0.0:5900
28/03/2021 23:53:51 Clearing securityTypes
28/03/2021 23:53:51 Advertising security type: 'TLS' (18)
28/03/2021 23:53:51 Clearing securityTypes
28/03/2021 23:53:51 Advertising security type: 'TLS' (18)
28/03/2021 23:53:51 Advertising authentication type: 'No Authentication' (1)
28/03/2021 23:53:51 Re-binding socket to listen for VNC connections on TCP port 5900 in (all) interface
28/03/2021 23:53:51 Listening IPv6://[::]:5900
28/03/2021 23:53:51 Listening IPv4://0.0.0.0:5900
28/03/2021 23:53:51 Clearing securityTypes
28/03/2021 23:53:51 Clearing authTypes
28/03/2021 23:53:51 Advertising security type: 'TLS' (18)
28/03/2021 23:53:51 Advertising authentication type: 'No Authentication' (1)
28/03/2021 23:53:51 Advertising security type: 'No Authentication' (1)
28/03/2021 23:53:52 [IPv4] Got connection from client esp32-arduino
28/03/2021 23:53:52   other clients:
28/03/2021 23:53:52 Client Protocol Version 3.7
28/03/2021 23:53:52 Advertising security type 18
28/03/2021 23:53:52 Advertising security type 1
28/03/2021 23:53:52 Client returned security type 1
28/03/2021 23:53:55 Pixel format for client esp32-arduino:
28/03/2021 23:53:55   16 bpp, depth 16, little endian
28/03/2021 23:53:55   true colour: max r 31 g 63 b 31, shift r 11 g 5 b 0
28/03/2021 23:53:55 rfbProcessClientNormalMessage: ignoring unknown encoding type -313
```

llandsmeer, 2021-03-28:
The unknown encoding type error might be misleading, connecting to `vino-server` with another VNC client (which works 100% fine), I get this in the server logs:

```
29/03/2021 00:14:19 rfbProcessClientNormalMessage: ignoring unknown encoding type 9
29/03/2021 00:14:19 rfbProcessClientNormalMessage: ignoring unknown encoding type -65527
29/03/2021 00:14:19 Using compression level 9 for client localhost
29/03/2021 00:14:19 Using image quality level 1 for client localhost
29/03/2021 00:14:19 rfbProcessClientNormalMessage: ignoring unknown encoding type -131072
29/03/2021 00:14:19 Enabling NewFBSize protocol extension for client localhost
29/03/2021 00:14:19 Enabling LastRect protocol extension for client localhost
29/03/2021 00:14:19 rfbProcessClientNormalMessage: ignoring unknown encoding type -131071
29/03/2021 00:14:19 rfbProcessClientNormalMessage: ignoring unknown encoding type -131070
29/03/2021 00:14:19 rfbProcessClientNormalMessage: ignoring unknown encoding type -131069
29/03/2021 00:14:19 rfbProcessClientNormalMessage: ignoring unknown encoding type -309
29/03/2021 00:14:19 Enabling cursor position and shape (rich encoding) updates for client localhost
```

llandsmeer, 2021-03-28:
I just tried using x11vnc. While I get the same unknown encoding type, I do get `draw_area()` calls so this looks like a vino-server problem

Links2004, 2021-03-29:
-313 is:
https://github.com/Links2004/arduinoVNC/blob/7432977b7af01d7b0536782c5cc8083bc22aa571/src/rfbproto.h#L497
which is a TightVNC extension, which is supported by many servers today.

but it is currently not used by the code:
https://github.com/Links2004/arduinoVNC/blob/7432977b7af01d7b0536782c5cc8083bc22aa571/src/VNC.cpp#L215
so no problem there.

why vino-server is not working is a good question.
you can try to enable debug output may it gives some hints
https://github.com/Links2004/arduinoVNC/blob/7432977b7af01d7b0536782c5cc8083bc22aa571/src/VNC_config.h#L70-L101

llandsmeer, 2021-03-29:
Thanks, that explains the error. I'm already running with debug messages and the strange thing is that I don't get any message of something being wrong on either sides. But x11vnc works fine, and I was able to get the M5Paper to display the top left corner of my screen at a slow 2Hz, so I'm very happy :). I guess I'll just stick to x11vnc as vino-server was discontinued in 2016 apparently (why is that the default in ubuntu 20???). Thanks again for putting this project online!

Off topic, do you know if it's possible to tunnel arduinoVNC through SSH? I suspect not, but maybe someone did this already?

Links2004, 2021-03-30:
SSH and VNC on the ESP will be too much for the ESP since SSH includes encryption, and the ESP is it was limit with VNC already.
2Hz is a bit low not sure why with the ILI9341 at 320x240 I where able to get ~20Hz.

you can terminate the SSH on a other linux machine and give the ESP only the VNC it you for example need to go via the Internet.

seisdr, 2021-08-09:
fix?

closed 2024-05-31 by TheNitek