Links2004/arduinoVNC #37: Image not refreshing completely on change.

Bolado, 2024-05-16:
I'm using a NodeMCU v3 with an ILI9341 display, I was able to make the touch works and displaying the image from the connected VNC machine, but on image change, artifacts appears, like it doesnt update correctly, I show on this [video](https://youtube.com/shorts/-68ukonrULU?feature=share) the issue, any direction to fix this issue would be welcome.

TheNitek, 2024-05-27:
Which encoding are you using?

TheNitek, 2024-05-27:
Oh and which code version? Latest release or master from this repo? If it is the release, maybe try the latest master

Bolado, 2024-05-27:
> Oh and which code version? Latest release or master from this repo? If it is the release, maybe try the latest master

I cloned the repo, haven't used the release as it was quite old compared to the state of the repo.

> Which encoding are you using?

I haven't changed the VNC_config.h to be honest, so I would guess it's trying to use those three below? Or how would I be able to set the encoding? I searched if was on server side, but xtigervnc has no arguments to set the encoding. 

```/// VNC Encodes
#define VNC_RRE
#define VNC_CORRE
#define VNC_HEXTILE```

TheNitek, 2024-05-27:
Whats the output in the serial monitor? 
you can uncomment the VNC_ZLIB line in the config file (ignore the comment that this is not implemented) and see if it changes the behavior

Bolado, 2024-05-27:
> Whats the output in the serial monitor? you can uncomment the VNC_ZLIB line in the config file (ignore the comment that this is not implemented) and see if it changes the behavior

I tried it, then it doesn't compile because of this error
```cpp
 #include "miniz.h"
```
then I try to include miniz and it fails to compile because of this error:
```cpp
VNC.cpp:1635: undefined reference to `tinfl_decompress'
collect2.exe: error: ld returned 1 exit status
```

TheNitek, 2024-05-28:
Oh sorry, zlib only works on the ESP32 (or at least is only tested there).

what is the output on the serial monitor? Maybe remove lines 99 to 103 in the config to get full debug output

Bolado, 2024-05-28:
> Oh sorry, zlib only works on the ESP32 (or at least is only tested there).
> 
> what is the output on the serial monitor? Maybe remove lines 99 to 103 in the config to get full debug output

Sorry for delay, on the serial debug to be honest a lot of the same, handlers for raw and hextile encoding:

```[_handle_raw_encoded_message] Pixel left: 112
[_handle_raw_encoded_message] ------------------------ Fin ------------------------
[_handle_hextile_encoded_message] ------------------------ Fin ------------------------
[_handle_hextile_encoded_message] x: 52 y: 5 w: 16 h: 7!
[_handle_hextile_encoded_message] ------------------------ Fin ------------------------
[_handle_hextile_encoded_message] x: 52 y: 5 w: 16 h: 7!
[_handle_hextile_encoded_message] ------------------------ Fin ------------------------
[_handle_hextile_encoded_message] x: 52 y: 5 w: 16 h: 7!
[_handle_raw_encoded_message] x: 52 y: 5 w: 16 h: 7 bytes: 224!
[_handle_raw_encoded_message] msgPixel: 112 msgSize: 224
[_handle_raw_encoded_message] Pixel left: 112
[_handle_raw_encoded_message] ------------------------ Fin ------------------------
[_handle_hextile_encoded_message] ------------------------ Fin ------------------------
[_handle_hextile_encoded_message] x: 52 y: 5 w: 16 h: 7!
[_handle_hextile_encoded_message] ------------------------ Fin ------------------------```

If you need something more specific , lmk

TheNitek, 2024-05-30:
VNC_FRAMEBUFFER is enabled?