Links2004/arduinoVNC #43: not able to connect with vnc server

Sayanbm, 2024-06-16:
hey I'm running tight vnc server  on my raspberry pi zero 2w with 32bit os,
so after uploading code the on esp32 with a ili9341 display and running vnc server with the command

 tightvncserver :1 -geometry 320x240 -depth 16 -name "ESP32-67935 :5901" -alwaysshared -s 0 -nocursor -broadcast

i got this on the screen 👇

[image]
 
then i tried this command 👇

tightvncserver -geometry 320x240

but i got same results, for referance it's working fine on my vnc viewer  👇

https://github.com/Links2004/arduinoVNC/assets/156459491/d8ce4f12-bbf6-4759-aaf8-53a699c74c10

TheNitek, 2024-06-16:
Whats the output on Serial?

Sayanbm, 2024-06-17:
> Whats the output on Serial?

Connecting to sayanbm
..
WiFi connected
IP address: 
192.168.5.44
[SETUP] VNC...
!connected
[rfb_connect_to_server] Connected.
[_rfb_negotiate_protocol] Server protocol: RFB 003.008
[_rfb_negotiate_protocol] used protocol: RFB 003.008

Authentication OK
[VNC-SERVER] VNC Server config - Name: sayan's ESP32-67815C :5901 desktop (sayan:1)
 - width:320      height:240
 - bpp:16        depth:16       bigendian:0     truecolor:1
 - redmax:31     greenmax:63    bluemax:31
 - redshift:11   greenshift:5  blueshift:0
[VNC-CLIENT] Supported Encodings:
 - ZRLE
 - Hextile
 - Zlib
 - RRE
 - CoRRE
 - Raw
 [VNC-CLIENT] Supported Special Encodings:
 - SetDesktopSize
 - CursorPos
 - ContinuousUpdates
 - compresslevel: 4
[VNC-CLIENT] Client pixel format:
 - width:320      height:240
 - bpp:16        depth:16        bigEndian:1    trueColor:1
 - red-max:31    green-max:63    blue-max:31
 - red-shift:11  green-shift:5  blue-shift:0
[rfb_set_desktop_size] setting desktop size to 320x240
vnc_connect Done.
!connected
[rfb_connect_to_server] Connected.
[_rfb_negotiate_protocol] Server protocol: RFB 003.008
[_rfb_negotiate_protocol] used protocol: RFB 003.008

Authentication OK
[VNC-SERVER] VNC Server config - Name: sayan's ESP32-67815C :5901 desktop (sayan:1)
 - width:320      height:240
 - bpp:16        depth:16       bigendian:0     truecolor:1
 - redmax:31     greenmax:63    bluemax:31
 - redshift:11   greenshift:5  blueshift:0
[VNC-CLIENT] Supported Encodings:
 - ZRLE
 - Hextile
 - Zlib
 - RRE
 - CoRRE
 - Raw
[VNC-CLIENT] Supported Special Encodings:
 - SetDesktopSize
 - CursorPos
 - ContinuousUpdates
 - compresslevel: 4
[VNC-CLIENT] Client pixel format:
 - width:320      height:240
 - bpp:16        depth:16        bigEndian:1    trueColor:1
 - red-max:31    green-max:63    blue-max:31
 - red-shift:11  green-shift:5  blue-shift:0
[rfb_set_desktop_size] setting desktop size to 320x240
vnc_connect Done.
!connected
[rfb_connect_to_server] Connected.
[_rfb_negotiate_protocol] Server protocol: RFB 003.008
[_rfb_negotiate_protocol] used protocol: RFB 003.008

TheNitek, 2024-06-17:
Can you als post the server log?

Sayanbm, 2024-06-18:
> Can you als post the server log?

16/06/24 14:17:27 Got connection from client 192.168.5.44
16/06/24 14:17:27 Using protocol version 3.8
16/06/24 14:17:27 Full-control authentication passed by 192.168.5.44
16/06/24 14:17:27 Pixel format for client 192.168.5.44:
16/06/24 14:17:27   16 bpp, depth 16, big endian
16/06/24 14:17:27   true colour: max r 31 g 63 b 31, shift r 11 g 5 b 0
16/06/24 14:17:27 rfbProcessClientNormalMessage: ignoring unknown encoding 16
16/06/24 14:17:27 Using hextile encoding for client 192.168.5.44
16/06/24 14:17:27 rfbProcessClientNormalMessage: ignoring unknown encoding -223
16/06/24 14:17:27 Enabling cursor position updates for client 192.168.5.44
16/06/24 14:17:27 rfbProcessClientNormalMessage: ignoring unknown encoding -313
16/06/24 14:17:27 Using compression level 4 for client 192.168.5.44
16/06/24 14:17:27 Disabling cursor position updates for client 192.168.5.44
16/06/24 14:17:27 rfbProcessClientNormalMessage: unknown message type 251
16/06/24 14:17:27  ... closing connection
16/06/24 14:17:27 Client 192.168.5.44 gone
16/06/24 14:17:27 Statistics:
16/06/24 14:17:27   framebuffer updates 0, rectangles 0, bytes 0

TheNitek, 2024-06-18:
Looks like your version of tightVNC does not like the setDesktopSize message. Try disabling it in the VNC_config by commenting the SET_DESKTOP_SIZE line out

Sayanbm, 2024-06-20:
Apparantly it didn't worked with tightvnc then i tried with x11vnc with help of it worked, but red and blue is interchanged

TheNitek, 2024-06-20:
In the meantime I changed the default config, so it should definitly work with tightvnc

Sayanbm, 2024-06-20:
It works but colour are changed this is how it looks ⬇️
https://github.com/Links2004/arduinoVNC/assets/156459491/9831ab31-6062-41bb-9b8b-50ccf85f9575
It the original ⬇️

https://github.com/Links2004/arduinoVNC/assets/156459491/4902032c-4af3-42b0-9455-d78c9c911209

TheNitek, 2024-06-20:
Is this x11vnc or tightvnc server?

Sayanbm, 2024-06-20:
> Is this x11vnc or tightvnc server?

It's happening with both of them

Sayanbm, 2024-06-26:
seems like those parameters can change the colures ,but shifting the values didn't quite worked for me ,also those came with some pre included values 

    opt.client.redmax = 31;
    opt.client.greenmax = 63;
    opt.client.bluemax = 31;

    opt.client.redshift = 11;
    opt.client.greenshift = 5;
    opt.client.blueshift = 0;

TheNitek, 2024-06-26:
Try
opt.client.bigendian = 0;

You can also try to use the last release instead of the latest master from this repo.

Sayanbm, 2024-06-26:
Thanks bro it worked!! Also thanks for helping me through out the entire process 🙏

closed 2024-06-26 by Sayanbm